Sienci-Labs/gsender #63: DRO's to small

cdo1955, 2021-09-10:
So far I really like everything about the software. Except for one issue. The DRO's are very small. The machine DRO is smaller yet and grayed out (or so it would appear). My favorite type of display shows machine and Works DRO's equal size, and a big size at that. Machine DRO's are just as important as Work DRO's especially when setting the machine.
Please make them bigger. Or make it an option to click on to go another tab that has bigger DRO's
Attached is my suggestion on how to size them up. Thanks for listening.
[image]

cdo1955, 2021-09-10:
Heres another suggestion

[image]

nmistry, 2021-09-21:
+1  -  Adding DRO readout size/font configuration makes this software more accessible to users with different display / eyesight constraints.

GoranP-se, 2023-03-16:
I use a touch screen monitor on my computer, 13.5". Even the "buttons" in this area very small for using a finger on screen, today I must use a pen for pointing...
Even the buttons for selecting Rapid/Normal/Precise are to small for using a finger..

It had been very nice if it where possible to move the barrier between this are and the visulizer to get a better possibility to use touch screen.

SienciLabs, 2024-11-06:
@cdo1955 @nmistry @GoranP-se we're underway on a gSender UI redesign that's currently available for download as Edge 1.5.0. You can see our discussion on it here: https://forum.sienci.com/t/gsender-edge-1-5-0-edge1-all-new-ui/14248/

Please check it out, check out our mockups, and let us know if our redesign will meet your needs for number visibility and button size

closed 2025-05-09 by kglovern