Sienci-Labs/gsender #64: Zero point when using Z-Probe is too low

nmistry, 2021-09-21:
Adding this issue from my [Reddit Comment](https://old.reddit.com/r/hobbycnc/comments/psmlgs/a_new_and_easy_grbl_gcode_sender/hdr4s5k/)   

> Z-Probing seems to be inaccurate. I set the depth to 14.09 (as measured by my calipers), run the homing, and then goto Z zero. The bit always digs into the material by about 0.8mm. This irregularity is not present when homing in UGS or Candle. Probably something I have misconfigured, but figured it was worth sharing.

This test was done on a SainSmart 3018 PROVer w/  grbl 1.1f and 1/8" flat nose end mill bit.    

Happy to share any additional details.

`$I
[VER:1.1f.20170801:]
[OPT:V,15,128]

$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]

$$
$0=10
$1=25
$2=0
$3=2
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=3
$24=20.000
$25=500.000
$26=250
$27=1.000
$30=10000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=500.000
$131=400.000
$132=100.000`

nmistry, 2021-09-21:
Adding a screenshot of the console after Z-probing.  

[image]

This is the height where the bit touches the work surface. 

[image]

Photo of the result.   

[image]

kglovern, 2021-09-29:
This should be fixed in the next build.  The probe module wasn't updating to use the correct plate thickness as entered in the settings - that's why the command was "G10 L20 P1 Z15" when it should be "G10 L20 P1 Z14.09".

Thanks for the report!

closed 2021-09-29 by kglovern