Sienci-Labs/gsender #68: Add ramp to surfacing utility

rchouinard, 2021-11-15:
The current surfacing utility implementation does a plunge to begin the operation, and between layer depths. This works fine for surfacing with square endmills, but won't work with common surfacing bits like [this one](https://www.whitesiderouterbits.com/products/6210) from Whiteside, which requires a ramp-in if the bit can't start outside of the work piece, as is the case with most spoilboards.

SienciLabs, 2022-04-05:
That's a fair point @rchouinard, we'll add it to our todo list

kglovern, 2022-10-20:
@rchouinard This change should now be available in the latest release with the surfacing updates.  All generated paths now ramp in rather than plunging in to the workpiece.

closed 2022-10-20 by kglovern