Sienci-Labs/gsender #73: Ability to run server-only on Raspberry Pi, connect from remote client

tillig, 2021-12-03:
I'm a LongMill owner using CNCjs with a Raspberry Pi for a few reasons:

- If the RPi dies due to dust or something else, I'm out $35.
- I can set up the USB connections, etc., one time and not disconnect/reconnect, so I know the setup is good - check connections are still up and go.
- I can work on designs and other tasks using my expensive computer without worrying about the running LongMill program dying, hanging up, or failing due to computer sleep issues.

I'd love to be able to switch to gSender but the ability to run the UI separately from the server and connect from computer to RPi isn't there yet. I read on Facebook that this feature was on the way, but I don't see anywhere it's being tracked.

I'm not an experienced Electron developer, but I've done my share of JavaScript/TypeScript. Looking through the code, it appears the simplest way to get this to work would be:

- Expose the ability to run the server at the CLI separately from the UI. The `server-cli.js` pretty much has this already, it's just not exposed through the main Electron app executable. There would probably need to be the ability to add a shared secret somewhere for remote access so you can ensure random things on the network aren't trying to control the CNC machine.
- Allow the UI to configure a remote server endpoint. _Ideally_ you could configure a list of them, so if you have multiple CNC machines, each with an RPi server, you could select the machine from the list. This would include the shared secret for remote access. "Local machine" could be a built-in option on the list.
- Have some sort of persistent setting so when you start up the UI it will auto-select the last used server.

For the folks running only local, all of the above combined should be zero effect - they wouldn't need to configure anything, the "local" server would be the default, and things would continue working. For the folks wanting to connect remotely, some instructions on how to start the server as a service on RPi with the shared secret and then configure the UI to connect would cover it.

That said, maybe there are already things in the works here, so that's just an idea.

okkays, 2022-04-17:
I know this ticket is from December, but this sounds really appealing, especially with the new AutoZero touch plate. I'd buy that in a second if I could use it with a headless gSender, but my current setup (also with cncjs) depends on a headless raspberry pi.

Since I'm not familiar with the code, could a maintainer comment on how complicated this would be to implement?

sepiroth887, 2022-08-17:
I think this already works, just launch gsender with options -H 0.0.0.0 and -p 8080  to set listen address and port. You can pick interface IP and port as needed and launch on raspi? (I only tested on windows and that worked fine for me)

closed 2022-11-21 by kglovern

tcurdt, 2022-12-01:
Since this is closed as completed I tried gSender 1.2.1 EDGE.

    gSender -p 8080 -H 0.0.0.0 --remote --allow-remote-access

But I am a little confused.
Despite `--remote` it still opens the UI.
And the `--allow-remote-access` does not seem to be required?

Is there any documentation on this yet, @kglovern ?

Cjkeenan, 2023-04-25:
I too am interested on the current status of --remote because anything newer than 1.2.2 EDGE is having issues on my RPi, but 1.2.1 and 1.2.2 do seem to work after some encouraging, but like you describe, still opens the UI. Is this what --kiosk mode is brought in version 1.2.4?