Sienci-Labs/gsender #74: Workpiece autolevelling

dzervas, 2021-12-06:
It would be amazing to have workpiece autolevelling, like Candle or bCNC do to be able to cut PCBs at a consistent depth.

happycncer, 2021-12-16:
Would also second this request. Autolevelling is becoming almost a basic requirement of cnc router software these days - not just for cnc work but also for other operations such as vinyl cutting and engraving processes, something I do regularly - along with all the more 'day to day' cnc operations.

SienciLabs, 2022-06-14:
Another similar request https://github.com/Sienci-Labs/gsender/issues/90

tmenet, 2022-08-02:
I too would like to see this implemented. Great for PCB etching or fine engraving in large pieces.

ModuloFS, 2022-09-11:
An easier solution would be a probing function like it is available in linuxcnc: https://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G38-probe There you can advice linuxcnc with a comment (PROBEOPEN filename.txt) in the gcode to store the probing data in a separate output file. In Freecad for example you can import such a file and do your modifications at the gcode: https://www.youtube.com/watch?v=dY8SIL1cdmA

aaronmgd, 2024-12-19:
Following. Would love this feature.

stefan63, 2024-12-26:
Has anyone tried this with gSender?
Is it possible to adapt?
https://github.com/kreso-t/cncjs-kt-ext

nuxeh, 2025-01-28:
+1, then i could feasibly switch from bCNC

stefan63, 2025-03-21:
Found this..
https://www.autoleveller.co.uk/

repentsinner, 2025-03-26:
Will give autoleveller a try, but having this built in somewhere would be amazing for engraving tasks without vacuum workholding..

PierreRamoin, 2025-05-18:
+1
Using an external tool for this would mean many operations (disconnect/reconnect/rehome machine, open file on both, etc.)
Would be absolutely game changer

stefan63, 2025-07-20:
Anyworkprogress done with this feature request?

chrismakesstuf, 2026-03-09:
I also wanted to mention that @krudoy currently has a pending PR for a height mapping tool that's pending, so perhaps it would be good to combine collaboration on that too? https://github.com/Sienci-Labs/gsender/pull/798

stefan63, 2026-09-18:
For now and untill this function is implemented in gSender I will vive Flatcam Plus a chance, it actually seams to do what I need.
https://github.com/thebestgoodguy/FlatCAM-Plus

tonedef, 2026-10-06:
Any progress on this? Can I help? I see PR [#923](https://github.com/Sienci-Labs/gsender/pull/923) is in draft. Pretty much a hard requirement for milling most PCBs. gSender looks phenomenal, this is literally the only feature that prevents me from using it.

stefan63, 2026-10-07:
> Any progress on this? Can I help? I see PR [#923](https://github.com/Sienci-Labs/gsender/pull/923) is in draft. Pretty much a hard requirement for milling most PCBs. gSender looks phenomenal, this is literally the only feature that prevents me from using it.

I agree, and an earlier comment to do this as a plugin would be nice!