Sienci-Labs/gsender #89: RaspberryPi issues running gsender

dosch, 2022-05-14:
I have two ways of running gsender on my Pi3 with Raspian Bullseye.

1. Using the [AppImage](https://github.com/Sienci-Labs/gsender/releases/download/v1.0.7/gSender-1.0.7.AppImage).
- I downloaded the AppImage file using the build in Chromium browser
- Moved the downloaded file to a dedicated Applications directory in my home directory
- Right-clicked the icon to view the properties
- Changed the execute properties under the Permissions tab from Nobody to Everyone

When I double click the icon the OS dialogue asks me if I am sure I want to run this executable.
I click Execute: Nothing... No feedback, not warning, no output. Just nothing as if nothing has happened.

2. Using the [ARM64 deb package](https://github.com/Sienci-Labs/gsender/releases/download/v1.0.7/gSender_1.0.7_arm64.deb).
- I downloaded the deb package.
- Installed it using `sudo dpkg -i gSender_1.0.7_amd64.deb`
- The gsender application appears under 'Utilities' in the app menu.
- When I open it, for a while nothing happens but then the splash screen is shown.
- The splash screen never disappears anymore and in the task bar a task is shown named 'Splashscreen.gif'

tnsondag, 2022-05-14:
I have the same issue.  Running from the command line, I get errors about not being able to open the .node file for “@serialport/bindings”.  This file exists with the correct permissions, but when I run “file” on it, it appears to be for x86_64 instead of aarch64.  

I tried replacing the “bindings” directory with one I installed myself via npm, but then it just segfaults (maybe a difference in version? I have no experience with node).

dosch, 2022-05-16:
I got version 1.0.5 to work. I downgraded to Debian Buster and then it worked

Dakyleman, 2022-07-27:
I have the same issue with getting stuck on the splash screen, and am using Raspbian Buster OS. 

1.1.2,1.1.1, and 1.1.0 have the identical issue of being stuck on the splash screen, both with AppImage and with the deb package.

Norm310, 2022-07-27:
I also have the same issue with 1.1.2,1.1.1, ...getting stuck on the splash screen, I tried from a newly installed OS, and did not work.

tindalld, 2022-08-08:
was there a fix for this. getting stuck on the splash screen on a RPI 64 bit clean OS.

Dakyleman, 2022-08-12:
@walidkayhan @kglovern @sophiabeluli 
Not pestering, but want to make sure this doesn't get buried. Would love to test the new versions on Pi :)

Dakyleman, 2022-08-17:
1.1.3 working great for me. Thanks!!!!

Cjkeenan, 2023-04-21:
Has anyone tried using the edge builds on bullseye? I am running into an issue where is says:
```
pi@rpi2b-cncjs:~/gsender $ ./gSender-1.2.5-EDGE-linux-armv7l.AppImage -- --remote --port 8082
/tmp/.mount_gSende5DIBoi/gsender: error while loading shared libraries: libatk-1.0.so.0: cannot open shared object file: No such file or directory
```

Any ideas?