Sienci-Labs/gsender #91: [Feature Request] Machine profile setting adjustment

gwiethaus, 2022-05-17:
Sorry about my English
In the settings, on the General tab > Machine Profile > Presset:
When choosing "Generic CNC" you should release the With, Depth and Height fields for manual filling in "Cutting Area from EEPROM"

SienciLabs, 2022-06-14:
Hi Guilherme,

The Width, Depth, and Height fields are all editable in the 'Firmware' tool or the console and aren't informed by the presets or the values at that location in the menu. You'll be looking for the $130, $131, and $132 settings. We just display those settings in the menu for your convenience.

Chris

gwiethaus, 2022-06-15:
Thanks Chris for replying. Now I understand from your explanation how the feature works.

Grateful.

Guilherme Wiethaus

Enviado do Email<https://go.microsoft.com/fwlink/?LinkId=550986> para Windows

closed 2022-06-27 by SienciLabs