Sienci-Labs/gsender #114: How to Internationalize?

kmimax, 2022-07-27:
Have you considered worldwide applications?

SienciLabs, 2022-09-16:
Yes, in fact before all the work that's been put in CNCjs - the codebase that we built off of - had internationalization in place. Our localization has definitely been lacking since there was so much we had to drop to prioritize a whole bunch of changes and features that we felt were missing. Even after working on gSender for coming up on 2 years now it still feels like we're only halfway to where we want it to be. Hopefully at the start of next year we'll have a chance to put the infrastructure back into place so that gSender can start to be used outside of English-speakers

SienciLabs, 2022-10-25:
To update this issue: due to the contributions of a community member, a localization spreadsheet has been started and is mostly complete for Spanish. Internationalization still needs to be implemented though and our spreadsheet needs to be turned into a json file so that other contributions can be made via github. Hopefully the process will continue to the end of this year