Sienci-Labs/gsender #118: Test Run button resets workspace to G54

tmenet, 2022-08-02:
If I am in any workspace above G54 and I perform a "Test Run", the test run changes the workspace back to G54 after it completes. If I do not notice and reset it back to my intended workspace, my job starts in the wrong workspace. This is not good... I do not expect test run to change my workspace. This could cause a disaster with my material if I have a special JIG placement.

Also wondering why does Test Run operate at hyper speed? Took me a while to realize you were animating the test.

kglovern, 2022-08-10:
GRBL check mode ($C) performs an automatic soft reset when it toggles off - that is what is resetting your workspace.  This is a firmware feature and not something that can be changed by the sending software.

https://github.com/grbl/grbl/issues/992

Test run runs as fast as the firmware can validate the lines since it doesn't have to wait for movement.  Most users don't wouldn't want it to run slower since the entire purpose is to verify that the lines will be interpreted correctly by GRBL.

tmenet, 2022-08-10:
Thanks for the explanation. I wasn't aware grbl did all of that. Perhaps you could add a pop up window warning that workspace origin will be changed  to G54 along with a checkbox to not show in the future would save someone from ruining their workpiece. I

tmenet, 2022-08-10:
I had another idea for a workaround. You could perform a $G before sending the $C to grbl and parse out and save the current workspace from the response. After the $C completes you could then output the saved G5x code that was originally set. Even if grbl fixes this in the future your code should continue to work.

kglovern, 2022-08-12:
Next release (1.1.3)should restore WCS modals after test run, similar to how Stop File restores WCS after reset 👍

tmenet, 2022-08-13:
👍 sweet.

tmenet, 2022-08-30:
Sorry to bring this issue up again, but as long as there are no errors in the gcode file, your fix worked perfectly! However when test run detects there is an error in the gcode file, the workspace is reset to G54.
 (usually this happens when a large workpieces is shifted and a line goes out of bounds.)

SienciLabs, 2024-11-05:
@tmenet Is this something you're still seeing in more current versions of gSender - like the recently released 1.4.10?

tmenet, 2024-11-08:
> @tmenet Is this something you're still seeing in more current versions of gSender - like the recently released 1.4.10?

I am running 1.4.10 and I have both hard limits and soft limits enabled. Lets say i select G55 and my Y zero is too close to the edge of the router. If I do an outline command it will cause an Alarm (2) and it will reset the workspace to G54.  I could not find a file that caused an error when doing a verify but getting any error still resets to G54.  I think the key to reproducing this is to have hard and soft limits enabled.

SienciLabs, 2024-11-08:
Ok, and just to clarify as well this is with the LongBoard running grbl firmware correct? It might be an issue of the firmware where we might have to look into reverting back the workspace manually on gSenders side of things

tmenet, 2024-11-09:
Yes, the original longboard running GRBL. Yes I think you are right and GRBL is resetting to G54 when it spots any error. Seems like if Gsender remembers the workspace you are in and GRBL changes it or you get an error, you make gsender revert back the original workspace.