Sienci-Labs/gsender #142: Homing Direction Invert firmware has incorrect options

Kerman, 2022-08-17:
The screen for invert home switch direction is incorrect, and it is impossible to set home switch directions.  

[image]

Kerman, 2022-08-17:
this leaves you in a 'broken' state where you cannot clear the homing violation to access the firmware settings to disable homing.  (I had to tape a wrench to the sensor to trip it so i could disable homing)

kglovern, 2022-08-17:
Thanks for reporting - we'll get it fixed in the next release.

In the meantime, for others having the same issue you can unlock the homing alarm using the unlock button in the top left of the visualizer and change the value in the gcode console manually over using a wrench.

[image]

kglovern, 2022-08-26:
This will be fixed in 1.1.4 releasing later today.  Thanks for the report!

closed 2022-08-26 by kglovern