Sienci-Labs/gsender #143: Add support for FluidNC

ModuloFS, 2022-08-19:
Hello,

i am using FluidNC as GRBL controller because of the better hardware performance compared to an Atmega. I have some problems to use gsender together with FluidNC. For example if i hit my E-Stop button, it leads to a "Door Open" detection inside my GRBL controller. To get out of this state i have to be able to reset the GRBL controller. At CNCjs for example there is a "RESET" button. I think it sends ctrl + R to the controller. This button is missing at gsender. 
Because of that i am asking you, would it be possible to support FluidNC (https://github.com/bdring/FluidNC) official by your software?
It would be great because your software looks pretty cool and has some nice features like the Jogging for example!

boboxx, 2022-08-23:
In would also like to see this added :)

cotepat, 2022-08-24:
Reset shortcut is mapped to Shift-5 (by default I think)... It's even faster than a UI button!

Also I have a workaround for (almost full) FluidNC compatibility (homing, soft limits, laser...). In fact, I tried to modify gSender but I gave up... In the end it was much easier to change FluidNC because the $ settings syntax is not supported by FluidNC and gSender is using that a lot! I created my own version of FluidNC where I added every $ settings used by gSender. These settings are fake settings that (do nothing other than being settable and queryable), but they mimic Grbl correctly. All you have to do is to configure all the $ settings once  (they are not linked to real FluidNC  values) and they are persistent afterwards.

It took very few changes to FluidNC... It was worth it because now gsender works perfectly.

ModuloFS, 2022-08-24:
Thank you very much for your soft reset hint, i have to try it as soon as possible! Up to now the normal gcode sending functionality with probing and the macros would be ok for me.

matthewfallshaw, 2022-09-04:
> I created my own version of FluidNC where I added every $ settings used by gSender…

@cotepat, can you share?
(I [checked your FluidNC fork](https://github.com/bdring/FluidNC/compare/main...cotepat:FluidNC:main), but it looks like you didn't push your changes.)

ModuloFS, 2022-09-11:
Now i have tried gSender together with original FluidNC and everything works perfectly if you sync both with the soft reset under the shortcut Schift+5! I do not need the firmware settings changed by gSender because i can change them realy easy with the FluidNC config file.

cotepat, 2022-09-12:
@matthewfallshaw here it is (source only). : https://github.com/cotepat/FluidNC/tree/v3.6.0-fake-grbl-config

Remember you have to force those "fake" settings to mimic FluidNC settings. They will be read by gsender (required for homing and soft limits).

ModuloFS, 2022-10-20:
@cotepat do you still use FluidNC as grbl controller? Since the FluidNC version 3.6.3 / 3.6.4-pre1 togehter with gSender 1.1.4 i get motion interruptions during jogging with speeds over approximatelly 3000 mm/min (https://github.com/bdring/FluidNC/issues/678). Do you have the same problems?

cotepat, 2022-10-21:
Yes! Unfortunately...

boboxx, 2022-10-22:
I think there are a few issues with FluidNC pin configuration in the few latest build. I saw a few issues being discussed in the discord forums.

ModuloFS, 2022-10-22:
The problem with motion interruptions have clearly to do with gSender. If i disconnect gSender i can jog the machine with the webUI or with FluidTerm without any issues...
@cotepat do you have any workaround for the problem or any guess what could be the problem? Maybe you could comment to the issue at FluidNC (https://github.com/bdring/FluidNC/issues/678) too so that you can see that several people have this problem.

ModuloFS, 2022-10-27:
Problem seems to be solved with FluidNC V3.6.4-pre3.

ModuloFS, 2022-10-30:
If anybody has connection problems between gSender and FluidNC i can recommend the modification in gSender told by Mitch in this post: https://github.com/bdring/FluidNC/issues/678#issuecomment-1294077552

With this modification you can connect and disconnect to FluidNC without any issues. The soft reset, made by the shortcut Shift + 5, is also not needed anymore! I tested it at a linux machine and it works there as well, you just have to substitute the programm path "C:\Programm Files" with "opt\"

Here is the copied comment from Mitch:

> The reset behavior of gSender has problems. In my setup, it would not connect to FluidNC at all because the JavaScript serialport module that gSender uses opens the port in a mode that results in the ESP32 restarting into bootloader mode, where it is expecting to upload new firmware. I solved that by modifying gSender to explicitly issue the proper hard-reset-into-run-mode sequence. I will tell the gSender folks about this. Meanwhile, if you want to patch it yourself, here is an easy way to do it without having to do a full recompile of gSender:
> 
> Edit the file C:\Program Files\gSender\resources\app\server\index.js - to do so you may need to edit as Administrator, or change the permissions on the file so an ordinary user can modify it.
> 
> Round about line 10242 you should see the line:
> 
>         _this2.emit('serialport:open', {
> Just before that line, add this:
> 
>         _this2.connection.port.set({ dtr: false, rts: true });
>         setTimeout(function() {
>             _this2.connection.port.set({ dtr: false, rts: false });
>         }, 300);
> Save the file and restart gSender, and it should connect to FluidNC properly. At least that worked for me (and Bart too).

ModuloFS, 2023-04-12:
Unfortunately i am not a software developer and i am not so deeply involved in this project. I changed the way to connect with fluidNC now. I removed the transistors which are responsible for the reset function from the esp32 board. Now I synchronize fluidNC with gSender by pressing the buttons ALTGR + Q  (@), i think this is the shortcut for stop job. This works perfectly for me now.