Sienci-Labs/gsender #155: "Start from Line" feature goes DOWN into work piece and breaks bits

tmenet, 2022-08-26:
I ruined two boards and busted two bits before I figured this one out. When I tell gsender to start at a line number that I have carefully chosen to be a Z move well above the work surface, the bit plunges into the workpiece and starts moving then pops up to the location I selected and proceeds as expected.  I played with this feature without a bit and it would do it every time regardless of where I started.

The only thing unusual about this gcode is my zero reference is on the bottom corner of my workpiece. (this is because all my wood pieces are at a slightly different thickness.) 

If, on the other hand,  I use the start from Line feature on a project that has Z=0 on the top surface, of the work piece, the Z axis goes up jogs to the starting point and then comes back down and proceeds as expected.

It's as if your code always assumes a top Z=0 and and goes a fixed distance above Z=0 to move towards the starting point.  If there is wood above Z=0 it goes right through the wood as in my case. 

Below is the link to the file I had problems with today so you can see for yourself. It has been a bad day.

https://drive.google.com/file/d/1bQYR3vpehQv90_UQGlQspM1FWu4nzAlL/view?usp=sharing

kglovern, 2022-08-26:
The assumption on start-from-line is that the user is taking away material and is zeroed to the top of the piece itself, not the top of the spoilboard since that is the more common method for novice level woodworkers.

We'll look at alterations that take into account situations such as yours where the highest Z point will be above Z0 when moving into position for start from line - we'll aim to get this into one of the next releases.

tmenet, 2022-08-28:
I figured this case was not taken into consideration. I use Onshape and Kiri-Moto and the bottom zero was an option. I figured it would help prolong my spoilboard with different thickness wood pieces on the same file. I ended up UGS to push forward and finish the job. It handles it fine.

You may just be able to scan the file and note the highest Z value you read and assure you go at least that or a little higher if you know you can. 

Your target user may have been a novice user but the feature set of GSender, the start and error handling and the clean layout makes gsender a clear favorite over all the other G code senders out there.

Thanks for looking into it.

cotepat, 2022-09-12:
Zeroing on the top of the spoilboard is very important in some situations, for instance when you want to thickness level uneven stock.

kglovern, 2022-10-20:
We've put some changes in to address this in the late build (1.1.6) - the initial Z movement to move into position is now 10mm above the maximum Z height of the toolpath, rather than 10mm above Z0.

closed 2022-10-20 by kglovern