Sienci-Labs/gsender #163: Built in Outline function or homebrew Boundary Marco don't work.

Taruuk, 2022-09-04:
As the title suggests, neither way of performing a quick bounding box trace will work on my Shapeoko. Below is my code to used in previous g-code Senders. I would guess that my installation is having issues with g-code variables. Any thoughts or other common experiences?

G90
G0 Z10
G0 X[xmin] Y[ymin]
G0 X[xmax]
G0 Y[ymax]
G0 X[xmin]
G0 Y[ymin]

SienciLabs, 2022-09-16:
Hi @Taruuk just to clarify, was this script working before in a previous version of gSender? Or is it something that you've used in other g-code senders before and you're asking how to make it work in gSender?

kglovern, 2022-10-07:
Hi @Taruuk 

We'll have a fix for this in the next release - the bounding box variables will again be accessible with the correct values within macros.