Sienci-Labs/gsender #164: [Feature Request] Raise Z to safe height on pause

J-eremy, 2022-09-06:
I was running a surfacing operation and had to pause it for some reason, and noticed when you hit pause it literally halts right in place. Needless to say, the 1-inch surfacing bit left a nice burn mark on the material. An option to set that height would be nice but just raising it past z zero would be good.

SienciLabs, 2022-10-25:
That's an interesting idea!

I think though it may not be doable. The quirk of pausing in GRBL is that it can happen in two ways: either an instantaneous pause or a planned one (like the one we use for tool changing functionality). The reason why the tool changing pause is able to move around in the paused state is because we're able to plan for that pause to happen and clear the buffer when that point is reached. In contrast an instantaneous pause happens so quickly because there's no buffer clear so that removes the ability to send other g-code through (like a z-axis raise). 

We'll keep it in mind though as a feature for the future if another method comes to mind

J-eremy, 2022-10-26:
Hey thanks for responding, I think the feature I am requesting already exists at the firmware level, altho disabled by default for some reason. The safety door switch feature in config.h sets up the feed hold button as a door safety switch and when triggered stops the feed, turns off the spindle and parks the z height at a safe level. on resume it does the opposite with delays to allow the spindle to restart and continues where it left off.

What I will have to do is recompile grbl with that flag on, and see how gsender responds. 

Not sure why they would have a feature that awesome and leave it disabled by default, but reading through issues on github it seems they don't subscribe to the whole freedom mentality and seem to have a goal of protecting people from themselves, even at the cost of features.

Now my question is, before testing does the pause button on gsender trigger the same loop as the feed hold pin on the physical board, or is it done completely in software? Because if it's essentially triggering the standard feed hold feature it might just work as intended right out of the box and can be simple documented and enabled at the end users discretion.

kglovern, 2022-10-26:
It's a combination of software and feed hold.  It internally pauses the workflow and sender so code doesn't continue to be sent, and sends a feed hold (!) to the firmware which triggers the firmwares hold state.

J-eremy, 2022-10-26:
Awesome, so it sounds like there might be some hope that when I test that feature it may just work. Now I just need some free time and then ill report the results.

SienciLabs, 2022-11-11:
Let us know what you find! If we have a chance we'll give it a shot ourselves too

closed 2022-12-07 by kglovern