Sienci-Labs/gsender #171: connect via network connection

tcurdt, 2022-09-25:
[ioSender](https://github.com/terjeio/ioSender) allows to connect via network.

[image]

Would be great to connect to some boards that feature an ethernet connection (that is often more reliable than USB).

Are there any plans to also support a network connection?

kglovern, 2022-09-26:
It's something we've discussed and have in the backlog, but no immediate plans to implement.  We'll make sure to comment when we revisit this.

tcurdt, 2022-09-26:
Native support would certainly be nice.
But maybe `socat` could be used as work around in the meantime.

https://stackoverflow.com/questions/29207980/bridge-serial-ports-over-network

Maybe worth giving a try.

tcurdt, 2022-11-28:
I probably try over xmas.
Whether it works or not - I still wish there was native support.

Franckapik, 2022-12-19:
I never use socat but i tried this : 

On rpi with arduino and drivers on ttyS0 : 
` sudo socat -d -d tcp-listen:9000 pty,link=/dev/ttyS0,raw,echo=0`

On my desktop computer : 
`sudo socat pty,link=/dev/virtualcom0 tcp:192.168.1.71:9000`

Then i tried to connect with UGS or bCNC to the virtualcom0 port . It seems to connect but i have no communication back :/

tcurdt, 2023-04-23:
@Franckapik did you make any progress on this?

TBH I am not even sure what protocol this is so this could maybe be tested without hardware.

tcurdt, 2023-04-25:
I got this working in a null-modem setup.
I will provide the details once I have confirmed it's also working in the real world. (should be in about 3 weeks)

grantemsley, 2023-05-11:
I am also interested in this. I tried using ser2net on the pi side, and com2tcp on my windows machine (com2tcp is an ancient program, but seems to still work ok). I can connect to the virtual com port using other tools like putty, but gSender doesn't even enumerate it's virtual com port, even after giving it a normal name like COM7.

tcurdt, 2023-05-30:
OK, so the big problem here SIP on macOS.

While with `socat` I can create a virtual serial port that bridges over tcp, the system protection prevents me to create the port under `/dev` where gsender is looking for it. And I haven't found a way to make gsender look for it somewhere else.

So it would be great to either allow to configure a path to a serial port instead of just searching under `/dev`.
...or have native support port serial over tcp.

tcurdt, 2023-05-30:
I am not sure I read this correctly:

https://github.com/Sienci-Labs/gsender/blob/ad47239f8146f602e30e847a5a35f8e991392712/src/server/services/cncengine/CNCEngine.js#L229

but I thought I could maybe inject a port by adding

```
"ports":[{
	path: "/Users/tcurdt/foo",
	productId: "6015",
	vendorId: "1D50",
	manufacturer: "manufacturer"
}]
```

to the `~/.edge_rc`.

But somehow gsender fails to fully start when I do (with just an empty file for `foo`)

kglovern, 2023-10-14:
Just an update on this issue - we have an early implementation of sending over telnet in the latest Edge build.

tcurdt, 2023-10-14:
Fantastic news!
That's [v1.3.6-EDGE](https://github.com/Sienci-Labs/gsender/releases/tag/v1.3.6-EDGE)

closed 2024-03-27 by kglovern