Sienci-Labs/gsender #186: Verion 1.1.6 xyz probe causes Z to go crash down at the end

tmenet, 2022-10-24:
Crazy movement at the end of a probing cycle. Z alone works fine. XY probing  cause gantry to crash in positive x direction. XYZ cause  Z to crash towards probe (down) at he end. If end limits are set you jest get an error at the end.

kglovern, 2022-10-24:
Hi @tmenet - could you look at your EEPROM settings and let me know if $13 is enabled?  If it is, disable it and let me know if you see the same behaviour.  We found a missing unit modal in the probe code and want to verify that fix.

tmenet, 2022-10-24:
$13 was not enabled. (set to mm) I enabled it ( inches) and the problem went away. So yes, you are on to the fix. Thank you!

kglovern, 2022-10-27:
This should be fixed in 1.1.7.  Make sure to remember to turn off $13 if you don't need it enabled (you definitely don't).

closed 2022-10-27 by kglovern