Sienci-Labs/gsender #204: Remaining issue with joystick movement

tmenet, 2022-11-11:
It seemed that for me, my joystick issues had gone away. I am running version 1.1.7 and with soft limits enabled if I am running at RAPID speed all works fine. However if I run at NORMAL or PRECISE speed it does not move or runs in small bursts. 

If I disable soft limits all  speeds work just fine.

tmenet, 2022-12-06:
It has been over 3 weeks and there have been no comments.  Maybe I should elaborate. I have observed this issue on two computers and two different joysticks and the issue persists in all cases.

Re-stated, the issue is IF soft limits are enabled, then movement in X Y and Z axis is in small bursts only when tapped. Long presses result in no movement. This problem only happens only at NORMAL  and PRECISE SPEED. Strangely at RAPID speed it works fine.

walidkayhan, 2022-12-08:
Hey @tmenet,

First off, thanks for the feedback. I did notice some issues with jogging with soft limits that weren't limited to just the normal and precise presets, this problem from what I experienced didn't depend on the preset that was selected. After some digging, I noticed there was an issue with how we calculated the max movements, hence why we don't see the machine move at all if we hold the shortcut to trigger the continuous jog (you should see the travel limit exceeded error in the console).

I implemented a fix for that and it should be in the next edge release **(1.2.2-edge)** and definitely in the next main release.

Thanks again for pointing that bug out 😃

tmenet, 2022-12-10:
Great, thank you. I would guess that version 1.2.2 edge is built off of 1.1.7? Just want to make sure it has all the 1.1.7 features.

Another question, when there is one G code file with multiple tool changes. If you have it set to manual tool change and gsender stops for a tool change, is it normal that the joystick does not jog any more and you have to use the controls on the console? I was not expecting that.

tmenet, 2022-12-12:
Thanks @walidkayhan 

While I am thinking about it and I have your attention there was another anomaly that I experienced that I ended up giving up on, I was trying to set the homing function to a joystick button. I never got it to work So I assumed it was intentional. At first I tried a two button simultaneous press which worked great with the stop jog command. It did not work so I tried a single button and that did not home either. So I assumed you didn't implement it because of the potential ease of inadvertently homing with a joystick. 

If you did not intend to disable this, it would be nice to have the homing feature in the joystick. The double button press works well and is very hard to do unintentionally.

Thanks for being so responsive.

walidkayhan, 2022-12-14:
Thanks for pointing that out as well, we recently made changes to safely handle actions that you can set to a controller button combination. There seems to be a problem right now with a few of them simply not working, homing being one of them as you mentioned. We'll be investigating these issues and hopefully have a fix for them in the next iteration of edge 😃

Modern-Memories, 2022-12-14:
I had been using a knock of set of joy cons for about a year without issue and after this last update gSender no longer recognizes them. I have gSender access to Bluetooth and full disk access just incase that was the problem. I do have an iMac 2021 with an M1 chip if that matters. I'll see if I can dig up another remote to test. It just sucks because I hade the shortcuts down pack on the joy cons hahah

tmenet, 2022-12-19:
Thank you @walidkayhan

I am not sure if I am doing something wrong but gSender-Edge-Setup-1.2.2-EDGE.exe 151 MB file only displays a splash screen and gets stuck on my laptop. I download several times and re-installed with same issue. It works fine on a desktop computer in another room so I can't use it with my Longmill.

It is a Toshiba laptop with only 6 GB of RAM. is has a i3 processor and up tp date 64bit Windows 10. Any ideas to try other than buying more RAM?

Modern-Memories, 2023-01-02:
@tmenet I would start a different ticker for the smash screen issue. I see they did mark this as a bug so they have it on their to-do list .

tmenet, 2023-01-03:
@Modern-Memories I have resolved the issue by adding more memory to my laptop. Its an old laptop originally with Windows 8 and now running Windows 10,  Going up to 12Gb fixed the issue. Older version of EDGE ran fine. <=v1.2.1 on 6GB of ram.

tmenet, 2023-02-15:
I have downloaded EDGE version 1.2.3 and it seems the issue that I started this thread regarding joystick movement not working at NORMAL or PRECISE when soft limits are enabled has not been resolved yet? Just hoping this doesn't get dropped.

walidkayhan, 2023-02-15:
@tmenet yes it is still in our backlog, we were just experiencing a bit more difficulties with shortcuts as a whole and are currently making some noticeable changes to it. 

Once it is in our next version of edge (or even possibly the main version), I'll post a comment here for clarification and close this issue.

walidkayhan, 2023-03-07:
@tmenet happy to report this has been fixed in the latest 1.2.4-EDGE release, going to close this issue, but feel free to open it back up if you find that you are still having issues with the joystick jogging.

closed 2023-03-07 by walidkayhan