Sienci-Labs/gsender #232: Enhancement: Add ability to run a macro to handle tool changes

crpalmer, 2022-11-28:
I see in git that you have a toolchange_revamp branch started and I'm afraid that you'll likely want to stay with that instead of my alternative proposal which also gives you more automated tool changes.  But, there's no harm is throwing the idea out there...  In

[github.com/crpalmer/gsender/tree/toolchange-macro](https://github.com/crpalmer/gsender/tree/toolchange-macro)

I added another option for toolchanges: Macro (in addition to the existing Ignore, Manual or Code).  When a toolchange is triggered, this macro is executed and can use M0 to pause as needed to prompt the user.  For example, using a Shapeoko Pro with a BitSetter, my macro will move to predefined location and ask the user to insert a tool before probing the bitsetter to set Z and then resumes the program ([Probe Tool macro](https://github.com/crpalmer/shapeoko/blob/master/cncjs/Probe_Tool.txt)).

Pushing the functionality to a macro vs. having gSender generating the gcode itself has the distinct advantage that you can easily customize what happens and you don't have to try to parameterize everything in the universe.  You can still provide example macros to make it easy for a user to handle a simple toolchange.

The big disadvantage of the approach I took is that you have to edit macros to change the parameters (tool change location, probe location, feeds, etc).  That makes it much less user friendly.  To solve this problem, you could think about implementing some way to define all the global parameters and providing a friendly user interface for editing them.

kglovern, 2022-12-05:
Hi,

It's already possible for the user to run any number of macros using the "Manual" strategy.  We considered adding a specific macro run option, but the current workflow is a little more flexible since they aren't locked into a single macro and can pick and choose depending on their current need without changing preferences.

This also helps your listed disadvantage - since you can already have various macros set up and just run the relevant one.

crpalmer, 2022-12-05:
Manual is a perfectly acceptable option for users that need to use multiple macros or need more manual control over the toolchange.  Having a "automatically run a macro" option doesn't preclude a user still using the Manual mode.

In my case, I am a Shapeoko user and all modern Shapeokos ship with a tool length probe that is attached at a specific location on your machine.  Since I am always using the same probing device and it is always in the same location, I will always be using the same macro.  In honesty, I have never once run anything except my "Probe Tool" macro in a toolchange.  I find it very nice to not have to think and that my machine just prompts me for a tool.

For me, and I would expect any Shapeoko user who is making the transition from Carbide Motion to gSender, the automatic macro workflow is significantly more convenient and removes one potential user error (I have accidentally clicked on the wrong macro when using a tiny screen).

Here's a quick screencast of what the workflow looks like from a real job I ran:

https://youtu.be/KxQCQMD3X6A

Since the last change to my branch, I've run 20+ jobs without any problems and at this point, if this was something you'd consider merging, I can make a pull request.  Please let me know if you'll consider it or if I just need to maintain my own changes on top of gSender.

Thanks!

hambudi, 2022-12-13:
I would love to use this too, I also use a SO Pro and have to run manual macros for tool-change which is not a blocker, but kinda annoying.

I have messed up jobs by running the wrong macros as well....

marcboivin, 2023-11-18:
Just ordered a rapid change ATC, which uses a normal spindle to make tool change and this feature would be nice here to.

From what I understand, my current options would be : 
- Make a simple script that would change M6 to a predefined set of command and parse the g-code before opening in gSender
- Wait for the SuperLongBoard with GRBLHal and build a user plugin
- Beg the gSender contributor to implement automatic macros/tool change support

I think the much better and safer option would be to have automatic tool change supported by gSender, short of that, automatic macros would be nice. 

Till that time I'll make my script and share it in case somebody wants to go down this ill advised path 🥇

crpalmer, 2023-11-19:
I included a link in an earlier comment to a github branch of gsender that
has this implemented in it, if someone wants to take a look at porting that
forward to the latest version.  I'm not able to do so as I've switched away
from using gSender at this point.  The code in my github repo was used
heavily for a few months though with no issues.

marcboivin, 2023-11-24:
From what I gathered the code base in those sections changed significantly.

Not willing to undertake an adaptation of your code.