Sienci-Labs/gsender #252: Cannot connect to machine

gcormier, 2022-12-29:
Ubuntu 22.04LTS
1.2.2EDGE

I ran a job last week, worked OK. Today, when connecting, as soon as I connect (while "Connecting"), Z plunges downwards immediately.

Connected via screen `screen /dev/ttyACM0 115200` and issued `$H` without any issues - the machine homed. Killed terminal and entered gSender, and tried to connect - it just hangs at connecting.

This is in the system log
```
Dec 29 13:05:47 cnc gsender.desktop[2216]: 2022-12-29T18:05:47.077Z - #033[32minfo#033[39m controller:Grbl Emitting Sender
Dec 29 13:05:47 cnc gsender.desktop[2216]: 2022-12-29T18:05:47.085Z - #033[31merror#033[39m uncaughtException: Cannot read properties of undefined (reading 'parent')
Dec 29 13:05:47 cnc gsender.desktop[2216]: TypeError: Cannot read properties of undefined (reading 'parent')
Dec 29 13:05:47 cnc gsender.desktop[2216]:     at IpcMainImpl.<anonymous> (/opt/gSender Edge/resources/app/main.js:332:54)
Dec 29 13:05:47 cnc gsender.desktop[2216]:     at IpcMainImpl.emit (node:events:526:28)
Dec 29 13:05:47 cnc gsender.desktop[2216]:     at EventEmitter.<anonymous> (node:electron/js2c/browser_init:161:11014)
Dec 29 13:05:47 cnc gsender.desktop[2216]:     at EventEmitter.emit (node:events:526:28)
Dec 29 13:05:47 cnc gsender.desktop[2216]: 2022-12-29T18:05:47.089Z - #033[31merror#033[39m: uncaughtException: Cannot read properties of undefined (reading 'parent') date=Thu Dec 29 2022 13:05:47 GMT-0500 (Eastern Standard Time), pid=2216, uid=1000, gid=1000, cwd=/opt/gSender Edge/resources/app/server, execPath=/opt/gSender Edge/gsender, version=v16.14.2, argv=[/opt/gSender Edge/gsender], rss=170201088, heapTotal=31215616, heapUsed=27927160, external=2851815, arrayBuffers=0, loadavg=[2.04, 1.51, 0.76], uptime=509.95, trace=[column=54, file=/opt/gSender Edge/resources/app/main.js, function=null, line=332, method=null, native=false, column=28, file=node:events, function=IpcMainImpl.emit, line=526, method=emit, native=false, column=11014, file=node:electron/js2c/browser_init, function=null, line=161, method=null, native=false, column=28, file=node:events, function=EventEmitter.emit, line=526, method=emit, native=false], stack=[TypeError: Cannot read properties of undefined (reading 'parent'),     at IpcMainImpl.<anonymous> (/opt/gSender Edge/resources/app/main.js:332:54),     at IpcMainImpl.emit (node:events:526:28),     at EventEmitter.<anonymous> (node:electron/js2c/browser_init:161:11014),     at EventEmitter.emit (node:events:526:28)]
```

gcormier, 2022-12-31:
I think this "fixes" things

1. open gSender
2. Power on machine
3. Hit "Firmware Settings"
4. Hit "connect to machine"
5. Close firmware settings.

kglovern, 2023-01-06:
Thanks, we'll take a look

closed 2023-03-06 by kglovern