Sienci-Labs/gsender #256: Setting Export fails

marioc262, 2023-01-05:
In Edge version 1.2.2 export of settings fails with a  forbidden error.

Running on macOS, tried on 13.1 and 10.15.1
```
GET http://127.0.0.1:49601/api/events?_=1672930033082 403 (Forbidden)
H._end @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:210
H.end @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:210
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:37
fetch @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:37
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
c @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
_e @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
a @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
Oe.i @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
De @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Or @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Er @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Qu @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Oe @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
jr @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Ut @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Nt @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295 

Uncaught (in promise) A {req: H, xhr: XMLHttpRequest, text: 'Forbidden Access', statusText: 'Forbidden', statusCode: 403, …}
_e @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
l @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
Promise.then (async)
_e @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
a @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
Oe.i @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:295
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
De @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Or @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Er @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
(anonymous) @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Qu @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Oe @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
jr @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Ut @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
Nt @ app.5747cf5af03a2d5ffb7c.bundle.js?_=1670617084030:338
```

kglovern, 2023-01-06:
Thanks, we'll take a look 👍

kglovern, 2023-01-27:
This will be fixed in an upcoming Edge release, thanks again for the report.