Sienci-Labs/gsender #276: Tool Change Issue -- v1.2.3-EDGE

majerus1223, 2023-02-13:
When the tool change command is detected, you are promoted to save positions via a dialog. This works the first time, however if you run another job, with the same configuration and attempt another tool change the dialog closes completely. 

v1.2.3-EDGE -- Windows 10 tried on 2 machines

majerus1223, 2023-02-13:
One other addtional thing I noticed. Within the UI there is a reference to "PH" what does this mean? Is this expected, if not can it be changed to something more human? 
[image]

kglovern, 2023-02-15:
We'll take a look at the issue with multiple toolchanges.

PH is placeholder.  Since Edge is the beta equivalent version not everything is final including text/labels.  It will be a more human copy by final release.

kglovern, 2023-03-15:
Crashing on subsequent tool changes should have been fixed in the latest edge release

closed 2023-03-15 by kglovern