Sienci-Labs/gsender #292: Request: Show preview when using start from line.

darickc, 2023-03-08:
It would be nice to see visually where the line I want it to start from is in the cut process (already cut lines are greyed out in the visualizer).  If something happens while cutting and the machine resets for some reason, I might not know what line it got to.  But if I can see what has been cut then I can get it close.

SienciLabs, 2024-11-06:
@darickc this is a great idea that we're still thinking about implementing on new gSender Edge

kglovern, 2026-09-15:
1.7.0-Edge will have a g-code simulator feature, which lets you validate the progress at specific lines ahead of time independent of external tools or guessing.  

This should address this issue and as such I'll be closing it.

closed 2026-09-15 by kglovern