Sienci-Labs/gsender #293: With Soft Limit Enable enabled, Jogging with Joystick does not work.

tmenet, 2023-03-09:
Soft Limit Enable prevents joystick from jogging properly  at Normal and Precise speeds. Rapid speed joystick jogging works fine. 

 If you disable Soft Limits all joystick jogging works fine.

walidkayhan, 2023-03-09:
Which version of gSender are you running? We fixed this issue on our latest edge release (1.2.4)

tmenet, 2023-03-09:
Yes, I downloaded the latest 1.2.4 since I have been waiting for this fix. I just confirmed it reports version 1.2.4.

The only odd thing was that this time my old joystick shortcuts were remembered from the last  version[ i think in the past the old setup was erased and I had to start new. Let  me know if there is something i should try.

walidkayhan, 2023-03-13:
Hmm, that is strange, yes we did some work on the shortcuts attempting to persist through app updates now. Would you be able to send your gsender settings file as well as the diagnostics report? It would help us find an issue if there is one there.

tmenet, 2023-03-13:
https://drive.google.com/file/d/1-xxjl8h_0fF1GjpIrWekFf5_-v_9WW9y/view?usp=sharing
https://drive.google.com/file/d/1-xKjpzmZBJ18_c7tBC0md026MwQKq8qM/view?usp=sharing

Above are links to download my gsender and firmware settings. i searched but could not find how to create a diagnostic report. let me know how and i will supply.

walidkayhan, 2023-03-27:
We are still investigating this issue, it has something to do with how jogging is calculated when your preferred units are in inches I believe. We'll update you once we get a fix. Cheers

tmenet, 2023-03-31:
I tried changing my display units to metric hoping that would make my joystick work properly. However I had the same problem until I disabled my Soft Limits. I mention this in hopes you are not going down a new rabbit hole.

tmenet, 2023-06-20:
@walidkayhan Sad to see this issue opened in March is still not resolved in v1.2.0 . Am I the only person using soft limits  to prevent missed steps?

J-eremy, 2023-07-14:
> @walidkayhan Sad to see this issue opened in March is still not resolved in v1.2.0 . Am I the only person using soft limits to prevent missed steps?

This was a problem that was fixed back before the edge version ever appeared. When edge and the regular version were forked, the problem was in edge and still not in main. Now it seems that the features from edge (and code) has been merged into the main branch bringing the problem back.  As of right now, on the most current version 1.2.2 the machine wont continuously jog when holding the key-bind with soft limits on, as it keeps going out of range.

I'm not sure, but it feels like something happened with the development team and they split between edge and release, and might have a slight communication issue between the two teams.

And no its not just you using soft limits.

tmenet, 2023-07-14:
@J-eremy 

Thanks for an explanation on what happened. It would be real nice to know when your teams can resolve this once again, It seems that whoever solved the problem before could easily merge his changes into both development teams,

I want to repeat that the problem is not just with the Joystick. I have since discovered that the keyboard shortcut keys (up, down, right, left arrow keys) don't work when held continuously just like the joystick fails.

J-eremy, 2023-07-18:
> 

Here is the issue I opened August of last year. It was definitely resolved and is the same thing that is back. I did notice that whatever they did absolutely didn't make it into the edge version (which is why I never used it)
https://github.com/Sienci-Labs/gsender/issues/156

The last version I used up until I updated recently to the current NON EDGE version was 1.17
https://github.com/Sienci-Labs/gsender/releases/tag/v1.1.7

Try that version, and I bet you will find that the issue your having doesn't exist.

tmenet, 2023-07-18:
I actually have been using version version 1.1.7 and in my experience it was only partially fixed since it only worked properly at RAPID speed. I opened up that issue  [#204 ](https://github.com/Sienci-Labs/gsender/issues/204#event-8686790876) on November 11, 2022 and it was closed reported as fixed in edge 1.2.4 It was not, which is why I opened it back up in this post.  I was hoping to see some comments from @walidkayhan  @kglovern or anyone at @SienciLabs would take a look at this and sort it out, Like I said before it ALSO does affect keyboard shortcut keys right, left, up and down arrows as well. They don't work when held when soft limits are enabled.

I love this sender I just would love to have a wireless pendant function and it is perfect.

J-eremy, 2023-07-23:
Outlining the job also throws an alarm.

J-eremy, 2023-07-25:
> I actually have been using version 1.1.7 and in my experience it was only partially fixed since it only worked properly at RAPID speed. I opened up that issue [#204 ](https://github.com/Sienci-Labs/gsender/issues/204#event-8686790876) on November 11, 2022 and it was closed reported as fixed in edge 1.2.4 It was not, which is why I opened it back up in this post. I was hoping to see some comments from @walidkayhan @kglovern or anyone at @SienciLabs would take a look at this and sort it out, Like I said before it ALSO does affect keyboard shortcut keys right, left, up and down arrows as well. They don't work when held when soft limits are enabled.
> 
> I love this sender I just would love to have a wireless pendant function and it is perfect.

#156 is the original issue I opened and was resolved, then I re-opened because of the regression. There was a response, it should be fixed in the next release.

tmenet, 2023-07-25:
@J-eremy Good to know. Thanks!

walidkayhan, 2024-03-01:
I will be closing this issue for now, please feel free to open a new one if joystick jogging is still not quite working for you :)

closed 2024-03-01 by walidkayhan