Sienci-Labs/gsender #294: M1 ignored after M8 in start gcode

jdforsythe, 2023-03-10:
I'm using the Digital Loggers IOT relay for the Longmill with both the router and laser power supplies plugged into the NC outlets.

The plan was to have starting gcode:

```gcode
M8
M1 ; Turn on the router or laser
```

and stop gcode:

```gcode
M9
M1 ; Turn off the router or laser
```

This will remind me to either turn on the router (which I will know worked because it will start running) or press the reset switch / turn the key on the laser module (which I will know worked because the fans will start running).

The problem is, when I start a program, it just starts going and ignores both the M8 and M1 lines. If I add ANY other M code in between, it will catch the first of them and ignores the last one, e.g.

```gcode
M8
M6
M1 ; Turn on the router or laser
```

This starting gcode will turn on the IOT relay and show the tool change dialog.

```gcode
M8
M0
M1 ; Turn on the router or laser
```

This will turn on the relay and show the M0 dialog.

```gcode
M8
M1 ;  Turn on the router or laser
M1 ;  Turn on the router or laser
```

And this works as my original intention, turning on the relay and showing the M1 dialog once.

The stop gcode does not fail or require multiple M commands to work.

jdforsythe, 2023-03-10:
This is v1.7.7 running on Mac

kglovern, 2023-03-15:
This was a long-term bug that was fixes as far as we know -  we'll take a look to see if there was some regression there.

SienciLabs, 2024-11-06:
@jdforsythe I just tested this in gSender version 1.4.10 and it seems fixed now, do you see the same on your end?

jdforsythe, 2024-11-08:
Yes this appears to be fixed

closed 2024-11-08 by jdforsythe