Sienci-Labs/gsender #303: Feature Request - Lock zero buttons

jdforsythe, 2023-03-16:
Accidentally zeroing, which I've done from time to time when wanting to click a "Go to" button, can be fatal to a project.

It would be cool to have a switch that can be turned on once you've properly set your zeroes, so that the zero buttons are disabled until the switch is turned back off. This would prevent accidentally zeroing in the wrong position during a tool change.

jdforsythe, 2023-03-16:
I wouldn't mind trying to tackle some of these. Is there any documentation on the architecture of this app?

Windcrest77, 2023-06-21:
Good suggestion.

This lock button should include a lock out for the probe as well. Basically, a lock you can set once you have your XYZ all established. If in lock mode then none of the XYZ setting operations (including probe) work unless you re-open the lock. I made the mistake once of accidentally starting a probe operation by inadvertently pressing that hot key on my macro pad. The probe slammed into the work. To prevent that mistake again I now keep my probe shorted out all the time unless I'm using it.

kglovern, 2024-10-22:
A feature to prompt on zero was added some versions ago to prevent this situation - it should accomplish the same goal of making the use cognizant of accidentally zeroing.

closed 2024-10-22 by kglovern