Sienci-Labs/gsender #320: Automatic tool change in 1.2.4-EDGE fails on Shapeoko 4 XXL by hitting soft limit

raquo, 2023-04-10:
Hello,

In gSender EDGE preferences I selected automatic tool change with coordinates X=-2, Y=-862, Z=-86. Those are the machine coordinates of my BitSetter. I can jog the machine to those coordinates just fine.

I have also enabled soft limits in the firmware, and enabled "view soft limit warnings" in gSender visualizer preferences.

My safe height in gSender preferences is set to 0 as it was by default.

1. To test this, I jog my machine to the middle of X, Y, and Z axes, and zero all.
2. I load a new file (gcode file attached) and click "start job"
3. The machine moves to the top of z axis (limit switch light is on). 
5. The UI now wants me to click on "Run GCode: Save Positions and Modals" button
6. I turn the router off and click on that button
7. The UI now says "Running..." and immediately shows the soft limit alarm in the background. The machine does not move physically.
8. Unlocking the alarm does nothing. The job start UI is stuck in the "Running..." state even if I cancel the job, re-home, and start the job again – I am never shown that "Save Positions and Modals" button again. I need to quit the application to see it again.

The machine hits the soft limit because gSender issues g-code [G91 G21 G0Z5](https://github.com/Sienci-Labs/gsender/blob/v1.2.4-EDGE/src/app/wizards/automaticToolchange.js#L87) (to raise the gantry by 5mm) while the gantry is already is maxed out.

For comparison, after an M6 command, Carbide Motion issues G0Z-5 command in absolute coordinates, placing the gantry 5mm below its highest possible level, if I understand correctly (and that's what in fact happens).

--

First, here is the gSender console log:

```
gSender - [Grbl]
Connected to /dev/tty.usbmodem14101 with a baud rate of 115200

Grbl 1.1f ['$' for help]
client $$
[MSG:'$H'|'$X' to unlock]
$0=10 (Step pulse time, μs)
$1=255 (Step idle delay, ms)
$2=0 (Step pulse invert, mask)
$3=5 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=3 (Status report options, mask)
$11=0.020 (Junction deviation, mm)
$12=0.010 (Arc tolerance, mm)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=25 (Homing switch debounce delay, ms)
$27=3.000 (Homing switch pull-off distance, mm)
$30=1000 (Maximum spindle speed, rpm)
$31=0 (Minimum spindle speed, rpm)
$32=0 (Laser-mode enabled as spindle, boolean)
$100=40.000 (X-axis travel resolution, step/mm)
$101=40.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$110=10000.000 (X-axis maximum rate, mm/min)
$111=10000.000 (Y-axis maximum rate, mm/min)
$112=1000.000 (Z-axis maximum rate, mm/min)
$120=500.000 (X-axis acceleration, mm/sec^2)
$121=500.000 (Y-axis acceleration, mm/sec^2)
$122=270.000 (Z-axis acceleration, mm/sec^2)
$130=865.000 (X-axis maximum travel, mm)
$131=865.000 (Y-axis maximum travel, mm)
$132=95.000 (Z-axis maximum travel, mm)
ok
client $H
ok
feeder $J=G21G91 X-5 F3000
ok
client $J=G21G91 X-7.00 F3000
ok
client \x85
feeder $J=G21G91 X-20 F5000
ok
client $J=G21G91 X-31.05 F5000
ok
client \x85
feeder $J=G21G91 X-20 F5000
ok
feeder $J=G21G91 X-20 F5000
ok
client $J=G21G91 X-89.67 F5000
ok
client \x85
feeder $J=G21G91 Y-20 F5000
ok
client $J=G21G91 Y-12.20 F5000
ok
client \x85
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Y-20 F5000
ok
feeder $J=G21G91 Z-10 F5000
ok
feeder $J=G21G91 Z-10 F5000
ok
client $J=G21G91 Z-9.47 F4000.000
ok
client \x85
G90
G17
G21
G28 G91 Z0
G90
T1 (M6)
ok
ok
ok
ok
ok
ok
feeder G4 P0.5
ok
feeder M5
ok
feeder G4 P0.5
ok
feeder G91 G21 G0Z5
ALARM:2 (Soft limit)
[MSG:Reset to continue]
```

The g-code file of the job itself is attached. It's generated by Fusion with GRBL postprocessor.

[jig door-1.nc.txt](https://github.com/Sienci-Labs/gsender/files/11186796/jig.door-1.nc.txt)

kglovern, 2023-04-11:
Thanks, we'll take a look at changing this.

kglovern, 2023-04-24:
This will be changed in an upcoming Edge release, thanks for the report.

closed 2023-04-24 by kglovern