Sienci-Labs/gsender #341: EDGE builds failing to run on Raspberry Pi

grantemsley, 2023-05-12:
Using a Raspberry PI 4, with Buster 32 bit as recommended in the install instructions. gSender-1.2.2-Edge-armv7l.AppImage is the last version that successfully runs.

It looks like the build process changed after that to start using CI builds?

When running gSender-1.2.3-EDGE-linux-armv7l.AppImage (and any newer versions), I get this error:

```
09:05:57.254 › Error: /tmp/.mount_gSendeVKb85C/resources/app/node_modules/@serialport/bindings-cpp/build/Release/bindings.node: wrong ELF class: ELFCLASS64
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:2:1822)
    at Module._extensions..node (node:internal/modules/cjs/loader:1226:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:2:1822)
    at Module.load (node:internal/modules/cjs/loader:1011:32)
    at Module._load (node:internal/modules/cjs/loader:846:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1035:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at load (/tmp/.mount_gSendeVKb85C/resources/app/node_modules/node-gyp-build/node-gyp-build.js:22:10)
    at Object.<anonymous> (/tmp/.mount_gSendeVKb85C/resources/app/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:11:46)
    at Module._compile (node:internal/modules/cjs/loader:1141:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1196:10)
    at Module.load (node:internal/modules/cjs/loader:1011:32)
    at Module._load (node:internal/modules/cjs/loader:846:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1035:19)
09:05:57.265 › Returned - http://undefined:undefined
09:05:57.268 › Unable to start the server at http://undefined:undefined
```

closed 2023-06-08 by kglovern

tcurdt, 2023-11-08:
Not sure whether I should create a new issue - the title would be the same.

I have installed `sudo apt-get install ./gSender-Edge-1.3.7-EDGE-arm64.deb` on a RPi4

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

but when starting it I am getting

`gsender-edge 
[2649:1108/002816.785869:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files
`

Checking the install instructions it says:

> Buster 32bit or earlier
> Bullseye (the latest release) is both 64bit and lacking libraries that Electron requires to run so is not currently supported

Buster is `oldoldstable` and LTS support runs out in a few months.
So I guess it would be nice to re-visit this?
Or is this already possible with AppImage?

So then I tried the AppImage

```
$ chmod a+x *.AppImage
$ ./gSender-Edge-1.3.7-EDGE-arm64.AppImage 
./gSender-Edge-1.3.7-EDGE-arm64.AppImage: error while loading shared libraries: libz.so: cannot open shared object file: No such file or directory
$ ldd ./gSender-Edge-1.3.7-EDGE-arm64.AppImage 
	not a dynamic executable
$ sudo ldconfig -v | grep libz
	libzstd.so.1 -> libzstd.so.1.5.4
	libz.so.1 -> libz.so.1.2.13
	libz3.so.4 -> libz3.so.4
```

Does that mean there is a problem with the AppImage?
Does it work for somebody else?

tcurdt, 2023-11-10:
FWIW I also tried the deb on a Debian 11 system and it still fails to start.