Sienci-Labs/gsender #345: Feature Request - Bigger feed rate override buttons

Windcrest77, 2023-05-17:
Would you please put the feed rate override buttons in a more prominent place and make them bigger, especially the "++" and "--" buttons. IMO the feed rate override buttons are **the most important buttons in the program**, more important than the jog buttons even. Because the override buttons are used while cutting and you could be destroying material because the feed rate needs adjustment, and you only have mere seconds to grab the mouse and find those currently very small and grayed-out override buttons. By the time you get to them several inches of material could have been cut at a non-optimal feed rate. 

Having the feed rate "++" and "--" buttons pop up over the visualizer when you press Start and be big and red would be great! Currently they are way too small for such a critical function at a actual cutting time. Cutting time buttons should always be easy to get to after Start, the Setup buttons are fine, there is no panic or hurry when the machine is not running yet. But something as basic as feed rate override should be easy to get to fast.

kglovern, 2023-06-08:
In the next release, feed/spindle overrides are:

1) accessible before the job starts
2) Sliders for larger adjustment with buttons for smaller adjustments
3) Increased in size

Hopefully this addresses at least some of your concerns - specifically being able to start at a slower feedrate and adjust upwards from there before starting cutting should be a large quality of life improvement.

[image]

Windcrest77, 2023-06-19:
Excellent!!! I installed the new version on Friday. Bringing the overrides to a place of prominence in the UI makes it very easy to slow start my jobs and up the feed rate based on the chips I see. This is especially important when cutting aluminum, brass or copper where there is only a small window of optimal feed rate, which is best determined by looking at the actual chip size being thrown off. Wood has a pretty wide range of feed rates, you can be way off and still get good results, But with metal having the +/- feed rate buttons is a life saver, and slow start.

I have evaluated several hobby gcode senders to date and have found this project to be the best, I really like this program a lot!

kglovern, 2023-06-20:
Glad to hear it's working well for you

closed 2023-06-20 by kglovern