Sienci-Labs/gsender #367: Cannot open .gcode files in gsender 1.2.0

BruceASnyder, 2023-06-17:
gsender 1.2.0 will not open any gcode files.

BruceASnyder, 2023-06-17:
unistalling 1.2.0 and going back to 1.1.7

BruceASnyder, 2023-06-18:
Now gSender 1.1.7 will not start. All I get is the splash screen.

kglovern, 2023-06-19:
Hi, can we get a little more information on this?  What operating system/architecture are you using?  Are you connecting before loading, and if not have you tried loading before connecting?

If you want to revert, you'll likely need to rename your sender.rc file - you can find this in your user directory (for windows, C:/users/{your username}/.sender_rc - rename this to something else and see if 1.1.7 will start.

halofx, 2023-06-19:
I am seeing some weird behavior with 1.2 randomly deciding it will not load any files. Window pops, select the file, click open, and then it returns to the main screen like nothing happened. Just had the issue, tried multiple files that I know are good.  While troubleshooting, I disconnected, and then reconnected to the CNC and then I could load files.

BruceASnyder, 2023-06-19:
I uninstalled 1.2.0, renamed sender.rc and the reinstall of 1.1.7 work. Thanks.

halofx, 2023-06-20:
Ultimately I rolled back to 1.1.7 as well. I could connect to the machine, but not load files. I would disconnect, and load the file, but then have trouble reconnecting. If a file did load, and reconnect, sometimes it wouldn't start.

I like that I don't need to trip an end stop in 1.2, unlike 1.1.7 on my Genmitsu 4040 Pro, but ultimately that was less hassle than the loading files issue.

kglovern, 2023-06-22:
Should be addressed in 1.2.1, thanks for report.

closed 2023-06-22 by kglovern