Sienci-Labs/gsender #377: Feed rate override causes gcode errors

GibsT, 2023-06-27:
In version 1.20 and 1.21 I have found that using the feed rate override feature will cause gcode errors at random points in the file. 

To reproduce I can just load up a facing gcode file and increase feed rate override. It will happen at different points in the file and throw different errors. I have seen error 24,2, and 33. Seems to me like there is some sort of buffer issue. The file will also run fine if the feed rate override is not touched. 

The file will run fine on G-sender version 1.17 and other gcode senders when using feed rate override. 

[image]

I have attached the gcode as "face.txt"
[face.txt](https://github.com/Sienci-Labs/gsender/files/11876133/face.txt)

kglovern, 2023-06-27:
Thanks, we're aware and looking into it.

kglovern, 2023-07-06:
This should be addressed in the next release.

closed 2023-07-06 by kglovern