Sienci-Labs/gsender #391: Macro Variable addition not behaving consistently, Live 1.2.2 and Edge 1.3.0

AndyCXL, 2023-07-23:
I have a simple gSender macro, but simple addition seems to be giving issues - it concatenates the values as strings rather than summing them… gSender Edge 1.3.0 and gSender Live 1.2.2, both on Apple.

Macro to illustrate this, and the console log resulting follows. Machine is at X zero (G54) position to start;

; Manually set some variables, then sum them - this works
%VAR1 = -10.2
%VAR2 = -5.2
(Var1: [VAR1])
(Var2: [VAR2])
(Subt: [VAR1-VAR2])
(Adds: [VAR1+VAR2])
%VAR3 = VAR1-VAR2
%VAR4 = VAR1+VAR2
(Sub: [VAR3])
(Add: [VAR4])
; Use %posx system relative macro to populate variables, then sum them - this DOESN’T work
; The subtract works OK, the sum appears to concatenate the numbers as string.
%POS1 = posx
G0 X1
%wait
%POS2 = posx
(1: [POS1])
(2: [POS2])
%SUM = POS1+POS2
(Sum: [SUM])

Output when run:

ok
feeder (Var1: -10.2)
ok
feeder (Var2: -5.2)
ok
feeder (Subt: -4.999999999999999)
ok
feeder (Adds: -15.399999999999999)
ok
feeder (Sub: -4.999999999999999)
ok
feeder (Add: -15.399999999999999)
ok
feeder G0 X1
ok
feeder G4 P0.5
ok
feeder (1: 0.000)
ok
feeder (2: 1.000)
ok
feeder (Sum: 0.0001.000)
ok

There has to be a simple solution to this… but I cannot seem to figure out what! Why do variables not sum correctly if they were populated from the %posx system variable?
Tested in Live 1.2.2 and Edge 1.3.0 and get the same, curious result.

neilferreri, 2023-07-24:
Weird…
Can’t test now, but try:

...
%POS1 = Number(posx)
G0 X1
%wait
%POS2 = Number(posx)
...
as a workaround.

AndyCXL, 2023-07-24:
Thanks, Neil. Will take a look tonight or tomorrow (UK time) - adapting your originally authored macros to find the centre of an external feature, and a second to find the centre of a hole.
Will be interesting when PRB: switches on in Edge (due in 1.3.1) as I believe PRB.z is machine coordinate, so the relative mode/G54 adopted for %posz etc will need some thought. Part of the fun.

kglovern, 2023-07-25:
We'll be doing this automatically in the expression parser as of the next Edge release - any variables that look like numbers will be explicitly converted to the Number type before the expression is evaluated.

closed 2023-07-25 by kglovern