Sienci-Labs/gsender #395: Override ranges are not sufficient

rmeissn, 2023-08-02:
I am currently utilizing gSender as a beginner in CNC operations, and my level of experience is not yet sufficient to accurately determine the appropriate spindle RPM and feed rates from the outset. Consequently, I heavily rely on the override menu of gSender. However, I have observed that the override sliders are limited to a maximum of 200%. I am interested in surpassing this threshold, potentially by inputting a precise value (such as for RPM), rather than being constrained to adjustments in increments of 10% or 1% using the plus and minus buttons. An option might be a checkbox in the settings menu which enables to input custom values within the override tab.

For instance, I aim to identify the optimal spindle RPM for my task. On one occasion, I mistakenly initiated a job at 700 RPM instead of the 7000 RPM, and I could only elevate it to 1400 RPM (which sufficed for the task but sounded less than ideal). To address this limitation in the menu, my current approach involves setting a default of 5000 RPM, as this allows for a 200% increase (doubling) which matches the maximum RPM of my spindle, while also permitting lower settings when needed.

neilferreri, 2023-08-02:
This is limited by the grbl firmware, not gSender.
You can go as high as 255, but you'll have to update your firmware.
Although experience is valuable in determining feeds & speeds, so is math.  I suggest you read up on chiploads, and then compare that to others who have the same machine to get depth of cut suggestions.

rmeissn, 2023-08-07:
Oh, I see. I wasn't aware of that. Thank you very much for letting me know!
Yeah, dealing with chipload and math is proving difficult on my small and inexpensive machine. It's lacking the necessary power for the usual formulas. Others have recommended going full RPM, minimal DOC and higher feeds.

closed 2023-08-07 by rmeissn