Sienci-Labs/gsender #397: Remote Jog Control with Android-Device

hgkoenig, 2023-08-15:
I tried to jog my mill remotely with one of my elderly Android phones/tablets. But it always moves twice as wide as preselected. Also DRO counts twice as wide. If I use a second Windows PC as a Remote-Device Jogging always works correctly. I wonder, no one else has recognized and reported this problem. So I ask me, is this a bug or is is my fault?

kglovern, 2023-08-15:
Are you seeing two jogging commands sent in the console or just a single doubled value?

hgkoenig, 2023-08-15:
In the console I can see two jogging commands both with the correct value.

kglovern, 2023-08-21:
OK, it sounds like the button event is firing twice - maybe the onClick and onTouch are both firing if the device is indeed "elderly".  We'll take a look but it might be tough to replicate on our end.

hgkoenig, 2023-08-22:
Meanwhile I did some further tests:
Asus_Z01HD Android 8.0.0 -> Problem exists.
LG G4 Android 6.0 -> Problem exists.
Samsung SM-P605 Android 5.1.1 -> Problem exists.
I attached a USB-mouse via OTG-Adapter to the Samsung-tablet -> One sinle mouse-click fires also twice!
I used UGS-universal-g-code-sender  and its pendant ->jogging works correct with all 3 test-devices. 
I used OpenBuildsControl and its pendant -> jogging works correct with all 3 test-devices.
Accidentaly I dicovered this:
If I place and hold one finger elsewhere on a "empty" spot on the screen, I can jog correctly with a second finger with gsender remotely too. All tests work as described on Arduino  Uno with grbl and on STM nucleo F446 RE  as well.
I hope, my examinations will help you, to find a satisfying solution for this problem. Thank You!

kglovern, 2024-01-03:
This should be fixed in the upcoming release.

closed 2024-01-03 by kglovern