Sienci-Labs/gsender #409: edge arm builds

tcurdt, 2023-09-19:
Is there a particular reason why the edge builds don't have a macOS arm release asset?
Would be nice to have.

walidkayhan, 2024-02-22:
We'll be looking into adding macOS arm builds in the future for gSender, we ran into problems building it on our CI in the past but we've figured out the solution to handle it

closed 2024-02-22 by walidkayhan

tcurdt, 2024-03-03:
@walidkayhan That sounds like really good news - but I don't see any macOS arm builds listed here:

https://github.com/Sienci-Labs/gsender/releases/tag/v1.4.4-RC1

Do I need to look somewhere else?

hamanjam, 2024-03-03:
There have been a couple comments where they haven't focused as much on cross platform builds until they lock in a more "stable" release.

tcurdt, 2024-03-03:
@hamanjam Yeah, but "closed as completed" sounded like it has arrived also to the edge/rc builds.