Sienci-Labs/gsender #411: XYZ Probe doesn't take into account Touch Plate Length & Width

AlTaiRius, 2023-09-26:
In the settings under Probe where you define the Touch Plate parameters for a Standard Block, there are fields for Length & Width.  These fields aren't being used when probing on the XY axis.  In the src/app/widgets/Probe/index.jxs code the XY Movement appears to be defined by Line 561:
`let xyMovement = toolDiameter + 20; // 20mm + width of the tool`

It would be better if the probing pull back took into account the defined length & width so that you can probe anywhere on the XYZ Touch Plate and it will always retract far enough not to cause a problem.

The specific problem I'm having with a pull back of 20mm is that when you use any of the Chinese XYZ touch plates with a 14mm hole in the corner...it doesn't leave you much room to start your Z probe.  [https://ae01.alicdn.com/kf/H7a7dce1585eb438fb40cbae469488035c/Plug-and-Play-Precise-XYZ-Touch-Probe-CNC-Processing-Offline-GRBL-Mach3-Tool-Sensor-for-WorkBee.jpg_.webp](url)

kglovern, 2024-03-28:
The probe dimensions have since been removed.  It's unlikely that we'd ever create a probing algorithm that uses them because it would frankly be too slow for the average user.  

In this situation I'd probably recommend writing a macro to accomplish the same thing accounting for your unique touchplate dimensions.

closed 2024-03-28 by kglovern

duub, 2025-04-23:
It seems the version with the hole it's becoming an standard... i can't find a generic version of it for my diy cnc without the hole.