Sienci-Labs/gsender #421: EDGE builds failing on RPi (dietpi,Debian 12)

tcurdt, 2023-11-10:
I have installed `sudo apt-get install ./gSender-Edge-1.3.7-EDGE-arm64.deb` on a RPi4

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

but when starting it I am getting

`gsender-edge 
[2649:1108/002816.785869:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files
`

Checking the install instructions it says:

> Buster 32bit or earlier
> Bullseye (the latest release) is both 64bit and lacking libraries that Electron requires to run so is not currently supported

Buster is `oldoldstable` and LTS support runs out in a few months.
So I guess it would be nice to re-visit this?
Or is this already possible with AppImage?

So then I tried the AppImage

```
$ chmod a+x *.AppImage
$ ./gSender-Edge-1.3.7-EDGE-arm64.AppImage 
./gSender-Edge-1.3.7-EDGE-arm64.AppImage: error while loading shared libraries: libz.so: cannot open shared object file: No such file or directory
$ ldd ./gSender-Edge-1.3.7-EDGE-arm64.AppImage 
	not a dynamic executable
$ sudo ldconfig -v | grep libz
	libzstd.so.1 -> libzstd.so.1.5.4
	libz.so.1 -> libz.so.1.2.13
	libz3.so.4 -> libz3.so.4
```

Does that mean there is a problem with the AppImage?
Does it work for somebody else?

_Originally posted by @tcurdt in https://github.com/Sienci-Labs/gsender/issues/341#issuecomment-1800710304_

tcurdt, 2023-11-10:
I ran into two problems:
1. the [wrong binding to libz](https://github.com/Sienci-Labs/gsender/issues/420)
2. the system needs `fuse` to be installed for the `AppImage`

After working around this another problem shows up

```
$ ./gSender-Edge-1.3.7-EDGE-arm64.AppImage 
[1772:1110/165507.338754:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files
[1804:1110/165508.583759:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.584433:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.584902:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.585581:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.586078:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.586547:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.587055:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.590861:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.591693:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.592448:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.593238:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.594095:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.594779:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
[1804:1110/165508.598575:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
16:55:11.715 › Error: /tmp/.mount_gSendeyDpkto/resources/app/node_modules/@serialport/bindings-cpp/build/Release/bindings.node: cannot open shared object file: No such file or directory
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:2:2131)
    at Object.<anonymous> (node:internal/modules/cjs/loader:1356:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:2:2131)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
    at node:internal/modules/cjs/loader:967:12
    at Function._load (node:electron/js2c/asar_bundle:2:13327)
    at Module.require (node:internal/modules/cjs/loader:1150:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at load (/tmp/.mount_gSendeyDpkto/resources/app/node_modules/node-gyp-build/node-gyp-build.js:22:10)
    at Object.<anonymous> (/tmp/.mount_gSendeyDpkto/resources/app/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:11:46)
    at Module._compile (node:internal/modules/cjs/loader:1271:14)
    at Object..js (node:internal/modules/cjs/loader:1326:10)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
    at node:internal/modules/cjs/loader:967:12
    at Function._load (node:electron/js2c/asar_bundle:2:13327)
    at Module.require (node:internal/modules/cjs/loader:1150:19)
16:55:11.727 › Returned - http://undefined:undefined
16:55:11.730 › Unable to start the server at http://undefined:undefined
[1823:1110/165512.471875:ERROR:platform_shared_memory_region_posix.cc(214)] Creating shared memory in /dev/shm/.org.chromium.Chromium.6Lztpd failed: No such file or directory (2)
[1823:1110/165512.472759:ERROR:platform_shared_memory_region_posix.cc(217)] Unable to access(W_OK|X_OK) /dev/shm: No such file or directory (2)
[1823:1110/165512.473097:FATAL:platform_shared_memory_region_posix.cc(219)] This is frequently caused by incorrect permissions on /dev/shm.  Try 'sudo chmod 1777 /dev/shm' to fix.
```

`sudo chmod 1777 /dev/shm` did not help - but it seems like the first problem to solve is that a shared library is missing for the serial ports.

tcurdt, 2023-11-10:
Hm. I don't see any shared library that could be missing.

https://github.com/serialport/bindings-cpp/blob/main/binding.gyp#L52

tcurdt, 2023-11-10:
Seems like the native code is the wrong architecture

```
$ file resources/app/node_modules/@serialport/bindings-cpp/build/Release/bindings.node
resources/app/node_modules/@serialport/bindings-cpp/build/Release/bindings.node: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=a2c7e256fa65faf2f6a6b6d2752ab23a9d6e323d, not stripped
```

tcurdt, 2023-11-12:
Isn't this a/the problem?

https://github.com/Sienci-Labs/gsender/blob/master/scripts/electron-builder.sh#L20

It seems to do a `npm install` on the host platform - not the target platform.
I might be missing something - but that's how I read it.

tcurdt, 2023-11-12:
Hm. Maybe not. Seems like it should be built on the proper host.

https://github.com/Sienci-Labs/gsender/blob/master/appveyor.yml#L246

I guess I need some feedback here to dig deeper.

kglovern, 2023-11-13:
We haven't been releasing PI builds for EDGE.  They will resume shortly as we evaluate our CI options.

If you want a working version, you'll need to compile it yourself on the host platform.

tcurdt, 2023-11-13:
@kglovern thanks for the response.

I was under the impression the host system might not really matter with an AppImage - as long as the arch matches.
Did I get that wrong? I never used AppImages before.
Or could the kernel still make difference here?

Either way the x86-64 binaries in an arm image seem a little off to me.

Is there some documentation building? The CI configuration is probably the closest there is?

So building would mean running this on a RPi?

https://github.com/Sienci-Labs/gsender/blob/master/appveyor.yml#L247

closed 2024-03-27 by kglovern