Sienci-Labs/gsender #423: 1.3.7-EDGE Jogging via Joystick keep moving after button release

vazhure, 2023-11-14:
(Version 1.3.7-EDGE)
I use joystick to jog, but after I release the button it keep moving...

Suggestion: please append possibility to override feed rate and spindle rate to analog axes (I plan to use potentiometers)

walidkayhan, 2023-11-14:
When you say release the button, are you referring to an actual button or releasing the joystick/thumbstick?

vazhure, 2023-11-14:
Joystick button. I tried ps4 Joystick, and my own based on Arduino pro
micro. Same result. In the release version it works fine, but not works
with turned soft limits in grbl settings. Home all button sometimes
disappears.

walidkayhan, 2023-11-15:
Thanks for the clarification. We did notice this issue as well, a fix for it has already been created and it will be in the next EDGE version.

closed 2023-11-15 by kglovern