Sienci-Labs/gsender #427: Macros disappear on restart in v1.3.8-EDGE

vsergeev, 2023-11-21:
Macros disappear after program restart on v1.3.8-EDGE, on Windows standalone desktop app. They persist on v1.3.7-EDGE.

kglovern, 2023-11-21:
Could you check permissions on your edge_rc file (in ~/users/{your username}/ and double check you have read + write permissions?  We aren't able to recreate this with testing.

vsergeev, 2023-11-22:
@kglovern sorry, it was the keyboard shortcuts, not the macros. The config file didn't appear to have any permissions issues, but to be safe, I deleted it for a clean slate. When I added a macro, I noticed it was added to the object, and it did persist across restarts. When I define a keyboard macro, however, I didn't see anything change in the config file, and it was lost across restart.

Hope that helps.

walidkayhan, 2023-12-07:
This issue has been fixed in **v1.3.9-EDGE**, going to close this issue for now, feel free to open it back up if you are still experiencing difficulties with keyboard shortcuts :)

closed 2023-12-07 by walidkayhan

walidkayhan, 2023-12-11:
@vsergeev I appreciate you pointing this out, yes, I noticed this too and you should be able to use the soft reset shortcut in this scenario. This fix will be in the next release :)

vsergeev, 2023-12-11:
@walidkayhan would you guys be receptive to allowing a resume / cycle start key binding (default can be unbound)? I can make a PR after confirming it's unlocked in the next release. This feels a little lighter touch than a soft reset and is appropriate for this use case.

walidkayhan, 2023-12-13:
@vsergeev Yeah feel free to open a PR, we will take a look at it and if the changes in it meet our standards and is not causing any side effects it will be merged in