Sienci-Labs/gsender #429: Unlock behavior changed in v1.3.x-EDGE

vsergeev, 2023-11-21:
Some machines (e.g. Carbide 3D Nomad 3) using GRBL require a cycle start / resume (`~`) or soft reset (0x18, I believe) to wake up the GRBL controller after a door open alarm. The unlock button in gSender v1.2.2 sent resume `~`, but gSender v1.3.7-EDGE just sends unlock `$X`. CNCjs has separate buttons for Cycle Start (`~`), Unlock (`$X`), and Reset.

At the moment, there isn't an easy way to resume with gSender v1.3.7-EDGE without a custom macro. Perhaps the unlock mode should be configurable or additional controls can be added.

kglovern, 2023-11-21:
We'll re-evaluate the change but generally we're ok with the unlock button just sending unlock commands ($X).  In the meantime, you should be able to use the keyboard shortcuts to achieve the same behaviour without adding custom macros.

vsergeev, 2023-11-21:
Unfortunately I'm on a tablet, so keyboard shortcuts are a bit difficult.

vsergeev, 2023-11-22:
I'm still investigating, but it seems that Soft Reset doesn't work in gSender, while it does in CNCjs. Not sure why, because the code paths look identical in GrblController.js. But currently, there I can't find a way to recover from the door state on the Nomad 3 without manually sending `~` in the console (since a keyboard shortcut doesn't exist for cycle start / resume, and all macros are disabled in this state).

vsergeev, 2023-11-22:
@kglovern Could the Soft Reset keyboard shortcut actually be disabled when it's in the door state, like the jogging commands and amcros are? That would explain why I don't see it get sent in the console in this state.

closed 2024-11-06 by SienciLabs