Sienci-Labs/gsender #432: Joystick not working in Jog Mode (GRBL)

vazhure, 2023-12-07:
Joystick dont work in jog mode
Sender send only "\x85" command (Jog command appeared only once and moving continued after button release)
Only can change Jog mode: rapid, normal, precise using Joystick button
[image]

vazhure, 2023-12-07:
seems it works when $20 = 1...
GrblController.js line 1732
                _let jogFeedrate;
                if ($20 === '1') {_

walidkayhan, 2023-12-07:
Thanks for pointing this out, I've figured out the cause of the problem, it's this little bugger here:

https://github.com/Sienci-Labs/gsender/blob/c4a2e808177e14b09f373d28d9a5bd33dbf082c8/src/app/widgets/JogControl/index.jsx#L989-L992

That check is no longer needed so I removed it, gamepad button jogging should work moving forward but I will monitor its behavior before the next release :)

closed 2024-01-03 by kglovern