Sienci-Labs/gsender #433: Home button section not visible on start (GRBL)

vazhure, 2023-12-07:
Home button section not visible on start.
It appears only i go to Firmware menu
[image]

[image]

kglovern, 2023-12-07:
Which controller are you running that you need to send a query status on connection?  If the "??" eeprom query isn't able to be sent on connection we're unable to determine whether or not to show the homing area.

vazhure, 2023-12-07:
> Which controller are you running that you need to send a query status on connection? If the "??" eeprom query isn't able to be sent on connection we're unable to determine whether or not to show the homing area.

MKS DLC 32

closed 2024-07-15 by kglovern