Sienci-Labs/gsender #436: Rapid override

Marty1109, 2023-12-19:
Anyway you can add a rapid override under the override tab. Be nice to start jobs at a slower pace to confirm movements at the start.

kglovern, 2023-12-19:
That is what the slider does - major adjustments are done using the slider, minor adjustments with the buttons.  It's able to be used before the job as is.

Marty1109, 2023-12-19:
i dont see a slide, what version was this added on? i only see a slide for feed rate and spindle speed overrides. nothing for rapids. i will add a screen shot later tonight when i am home.  believe i am on build v1.2.2

Marty1109, 2023-12-20:
[image]

[image]

SienciLabs, 2024-11-05:
@Marty1109 I see you're asking about rapid overrides - we never chose to make room for this in the gSender UI since it's fairly non-typical for hobbyists to use. I think if you're wanting it then the feedback is fair to give though, maybe sometime in the future we could look to fit it in somewhere 👍

closed 2024-11-05 by SienciLabs