Sienci-Labs/gsender #438: DRO - Simple Math

Marty1109, 2023-12-19:
Would be nice to see the DRO be able to do dimple math

ie. probe z axis (waste board top) subtract material thickness in DRO = theoretical material top for z

Coming from UGS just a few of the handy tools that gsender is lacking.. 

Running on a raspberry pi. Gsender is way more responsive and reliable. Keep up the good work

kglovern, 2024-01-02:
You are currently able to click the DRO WPOS and alter the current offset by entering it manually - the same way that UGS works.  Was there something specific you were missing that this doesn't accomplish?

Marty1109, 2024-01-02:
Hey Kevin,UGS was able to do simple math equations Say the position read 4.25” but you needed to change it by .25”… you could click on it and type -.25” and it would solve it.. Also i noted there is no rapid over ride to slow the rapid down… TroySent from my iPhone

SienciLabs, 2024-11-06:
#436 Rapid Override