Sienci-Labs/gsender #439: Spindle delay not added

hamanjam, 2023-12-20:
Spindle Delay on gSender 1.2.2 or Edge1.3.9 does not add the spindle delay to the gcode file based on the default grbl(mm) post processor in Vectric Aspire.  

Vectric post processor adds the speed and direction in a different format and gSender is unable to determine where to add the delay.  Setting a spindle speed of 27000rpm results in  gcode command "S27000M3" which is not in the expected format.

hamanjam, 2023-12-20:
Just installed Edge 1.3.10 and t does account for the formatting change but it only inserts a 1ms pause (G4 P1).  Here is my console output:

feeder m8
ok
T1
G17
G21
G90
G0Z6.350
G0X0.000Y0.000
M3 S22000 G4 P1
G0X29.739Y61.512Z2.540
G1Z-3.000F381.0
ok
ok
ok
ok

hamanjam, 2023-12-20:
Just installed Edge 1.3.10 and it does not account for the formatting change (I manually changed the cgode and it did add the G4 but it only inserts a 1ms pause (G4 P1).  Here is my console output:
...........

Not adding the delay:

feeder m8
ok
T1
G17
G21
G90
G0Z6.350
G0X0.000Y0.000
S16000M3
G0X100.000Y0.000Z2.540
G1Z-6.250F1016.0
G1Z-2.500
ok
ok
ok
G0Z2.540
...........

Added the delay but only 1ms:

feeder m8
ok
T1
G17
G21
G90
G0Z6.350
G0X0.000Y0.000
M3 S22000 G4 P1
G0X29.739Y61.512Z2.540
G1Z-3.000F381.0
ok
ok
ok
ok

hamanjam, 2023-12-26:
I asked for a solution on StackExchange and someone posted a regex that *may* work well.  Seems easier than what I was thinking at first.  I modified their suggestion a bit and I think it works.  It doesn't care if there is a space between anything with M or S first.  When the line is matched it adds the "G4 P1" *except* if the line has any G4 Px command.  What I have tested is:

 gcode.replace(/\b(?:S\d* ?M[34]|M[34] ?S\d*)\b(?! ?G4 ?P?\b)/g, '$& G4 P1')
 
 My question on Stack Exchange and the answers provided are here:
 https://stackoverflow.com/questions/77712131/javascript-replace-append-statement

hamanjam, 2023-12-26:
I've removed my comments on the forum gSender 1.3.10 release announcement.  I have varous tests with and without spaces and ordering of commands if necessary but I'm hoping my comment above and the StackExchange thread would provide useful n updating GrblController.js and GrblHalController.js

walidkayhan, 2024-01-09:
Thanks for the solution, we've replaced the old regex with this new one, it will be in the release 👍🏼

closed 2024-01-09 by walidkayhan