Sienci-Labs/gsender #440: Jogging 1.3.10-Edge

vazhure, 2023-12-20:
Seems Jog mode uses default values on the program start.
[image]

After reading GRBL settings from Firmware window it works for Z axis form me (calculates distance from position and max limit)
[image]

Other axes just starts from 16 mm, than every click on UI Jog button distance value multiplied by two. Initial value depends of Jog mode (Rapid, Normal, Precise). Should be calculated.

Log:
 _$J=G21G91 Z89.379 F800.000 // Calculated distance in desired direction by Axis Z - OK
ok
 \x85
 $J=G21G91 Y16.00 F1000 // First click on UI button
ok
 \x85
 $J=G21G91 Y32.00 F1000 // Second click on UI button
ok
 \x85
 $J=G21G91 Y64.00 F1000 // Third click on UI button
ok
 \x85
 $J=G21G91 Y128.00 F1000 // Fifth click on UI button
ok
 \x85
 \x85 // Not stopped_, stopped by pressing UI Button - Stop jogging

If I use Joystick it initially can jog continuously, but after reading GRBL settings in Firmware window it can work only by steps, even if I hold joystick button...

walidkayhan, 2024-03-01:
@vazhure has this issue been resolved for you in the latest version of gSender (v1.4.3)?

vazhure, 2024-03-02:
> @vazhure has this issue been resolved for you in the latest version of gSender (v1.4.3)?

yep, just tested. works fine now

walidkayhan, 2024-03-04:
Awesome, glad to hear that!

closed 2024-03-04 by walidkayhan