Sienci-Labs/gsender #441: Difficult to exit GRBL door state (v1.3.10-EDGE)

vsergeev, 2023-12-22:
Currently, there is no easy way to resume from the door state in some GRBL configurations, like the one used by Carbide 3D Nomad 3, because keyboard shortcuts and macros are disabled in this state, along with jogging and other controls, and the unlock button doesn't send the appropriate command to unlock GRBL. This wasn't an issue in gSender v1.2.2, because the unlock button sent resume `~` instead of unlock `$X`.

In order to resume from the door state in this GRBL configuration, either cycle start / resume (`~`) or soft reset (0x18) must be sent, not unlock (`$X`). Soft reset can't be sent because there is no button for it, the shortcut is disabled, and as far as I can tell it's not possible to type the escape character in the console (at least on Windows). Resume also has no button or shortcut, so the only option is to manually type and send resume with `~` in the console. This technically works, but is inconvenient for tablet users.

Some possible options:
1. Allow overriding Unlock button behavior in the settings to send either `$X`, `~`, or `0x1f`
2. Add a dedicated Resume / Cycle Start button like CNC.js
3. Allow sending a subset of keyboard shortcuts so that either soft-reset or a new resume shortcut can be sent to unlock GRBL

For Nomad 3 users, I think option 1 is best, since I don't believe `$X` has a use case with its GRBL configuration and we could repurpose the unlock button to send resume.

mattydboom, 2024-08-06:
I have noticed this same behaviour too on a SparkConcepts xProV5 controller running GRBL. Surprised this hasn't been addressed sooner than now. 

Plan on upgrading my controller soon though so I may end up working around that way instead. 

Good write up & suggestions though!

SienciLabs, 2024-11-06:
@vsergeev it's possible your request for this fix, specifically the #2 option you mentioned, should be implemented in gSender moving forward. Have you had a chance to check the new 1.5.0 Edge version of gSender. Let us know if the permanently available buttons are a resolution to your issue

closed 2025-06-17 by kglovern