Sienci-Labs/gsender #444: Feature Request: Suppress gcode comments in gcode sent to machines.

gsrunion, 2024-01-11:
As per https://community.carbide3d.com/t/so5-pro-inline-comment-firmware-issue/64097 there is a bug in the firmware for the Shapeoko 5 Pro that causes the machine to choke on inline gcode comments.

While that ultimately needs to be fixed on their end, forwarding gcode comments to machines serves no purpose. I am proposing a change in gsender that strips out any gcode comments as gcode is being sent to the machine.

kglovern, 2024-01-15:
We'll take a look at this for the next couple of releases.

gsrunion, 2024-01-15:
Awesome. Thanks. JavaScript isn’t my strong suite by I may take a stab at putting up a PR.

hamanjam, 2024-01-16:
I actually edited my post processor to add more comments such as file path and date saved  If you have no use for comments, it may be easier to edit your post processor to strip them out if they cause issues with your controller.  Parsing comments can be difficult.  Is there a specific Shapeoko post processor that doesn't produce comments?

For the purpose of stripping before sending, you'd have to regex to stop anything between "(" and ")" which would be something similar to :

string.replace(/\([^()]*\)/g, '')  

Similar thread with examples this solution was taken from:

https://stackoverflow.com/questions/640001/how-can-i-remove-text-within-parentheses-with-a-regex

gsrunion, 2024-01-16:
From what I understand the probing routines on the GSender side insert inline comments as well. https://github.com/Sienci-Labs/gsender/blob/d3c86520890b9580517c792977988382b91dffdf/src/server/controllers/Grbl/GrblController.js#L479 here is at least one example.

gsrunion, 2024-01-16:
> Interesting to see in "check" mode that it changes the commands to comments (I'm assuming just to negate sending the actual command when in debug) I could see that being a problem. I can't see the benefit sending comments to the controller on a regular basis when grbl basically ignores comments and we would never really see them.
>
> It sounds like your options are very limited to using propriety apps and file formats. Almost seems like it would be a lot of manual gcode editing to use anything outside their infrastructure. Learning curve and cost are the downsides but switching to something more "open" seems a bit easier but I totally understand wanting to use something that you're comfortable with. I looked at my Vectric post processor for Shapeoko and they only add their Vectric branding comments which would be easily removed. It's a shame you can't edit your pp in C3D. Hopefully you can get a workaround until the firmware is fixed (really surprised it's been around this long)

You have a few options if you pay for a pro version of Carbide Create. Otherwise you are stuck with the default. The do provide a tool to extract the gcode from their C3D files. But that is another step. The LightBurn folks are coming out with a CNC software. I have high hopes for that. If not I pony up to move from VCarve Desktop to pro to not be size limited.

gsrunion, 2024-01-16:
https://github.com/Sienci-Labs/gsender/blob/d3c86520890b9580517c792977988382b91dffdf/src/server/controllers/Grbl/GrblController.js#L418

and 

https://github.com/Sienci-Labs/gsender/blob/d3c86520890b9580517c792977988382b91dffdf/src/server/controllers/Grblhal/GrblHalController.js#L381

Might be the regex I would be looking for and, perhaps, the right place to do this replacement.

gsrunion, 2024-01-16:
https://github.com/Sienci-Labs/gsender/pull/448 I attempted the changes as per guidance found here. https://community.carbide3d.com/t/gsender-any-special-steps-to-use/72349/39

kglovern, 2024-01-25:
This should be merged and part of 1.4.1 release.  Thanks everyone for the feedback/pull requests.

closed 2024-01-25 by kglovern