Sienci-Labs/gsender #447: FEATURE REQUEST: Ability to send gcode via python script

J-eremy, 2024-01-12:
Hey I know this is a longshot, but a backdoor way to send commands through GSender to the machine via external scripting would be a fantastic addon to allow for much more complex scripts to be run for those who want to. 

An example would be the ability to add pretty much anything to the machine through external processing of scenarios.

Think of an advanced macro that the developers of GSender would not have to worry about or support, but will add insane amounts of options and or external controllers, Arduino based switch panels etc and so on.

Its actually amazing that something like this doesnt already exist in the software, like a rest api or something. Or maybe it does and im just not in the loop.

J-eremy, 2024-11-07:
Exciting that this was re-opened and feature and documentation tags were added. You guys really do rock. Thank you.

chrismakesstuf, 2026-04-06:
From @tillig on https://github.com/Sienci-Labs/gsender/issues/596:

> There are gcode entries for program events - start, stop, etc. but there's no way to integrate with an alert on those events, like an IFTTT webhook to alert me on program completion.
> 
> It'd be nice to have webhook support for program events so it can be integrated with other systems.

From @darickc:

> I created an application that will connect to gsender and publishes all the events to MQTT. You can run it locally, but made more to run in a docker container. I could modify it to add webhook capabilites if that is something that might be wanted. I'm going to be using it to hook into home assistant to turn on/off my dust collection when running a job. Check it out here: https://github.com/darickc/gsender2mqtt.

From @chrismakesstuf:

> @darickc are you at all related to this thread I came across recently here? https://community.home-assistant.io/t/mqtt-nodered-starter-assistance-please/917372/

From @darickc:

> @SienciLabs No. Maybe it is whitewolf from this thread: https://forum.sienci.com/t/homeassistant-integration/14111.