Sienci-Labs/gsender #449: Edge 1.3.10 Job list duplicates some entries

hamanjam, 2024-01-16:
I just noticed that most of my jobs are duplicated in the job table.  At first I couldn't confirm if they are in fact duplicates but they are.  I can't find a pattern right off but it seems like about 75% of the jobs I run show twice.    Also curious if it's possible to reset this information. 

[image]

kglovern, 2024-01-16:
We'll take a look - Are you running Verify/check job before the actual file?  Otherwise we'll see why it's getting sent twice.

We'll look at adding a way to clear history but that will be in a future update.

kglovern, 2024-01-16:
1) Verify is counting as a completed job for the purposes of both maintenance reduction and job stats when it shouldn't (no machine movement)
2) Double jobs seem to be reducing time twice as well

hamanjam, 2024-02-27:
I don't see the duplication in 1.4.3 (or previous 1.4.x versions)

closed 2024-02-27 by hamanjam