Sienci-Labs/gsender #451: gsender 1.3.10 EDGE must disconnect/reconnect after homing with Z limit switch activated

kbulgrien, 2024-01-20:
To replicate:

+ Open gsender 1.3.10 EDGE.
+ Connect to the machine.
  - In my case, the machine is set to **X-Carve 1000x1000**. it seems doubtful the issue is particularly machine-specific.
+ Click the **Click to Run Homing** button.
   - Allow the homing operation to complete.
+ Jog the machine into the Z-limit.
   - Note the **Alarm** condition and the **Click to Unlock Machine** button.
      * Do not click the **Click to Unlock Machine** button.
   - Note the **Home** button is not greyed out.
+ Click the **Home** button.
   - Note the **Homing** indicator.
   - Note the machine is not actually moving any axes.
   - Note that the UI is largely greyed and unresponsive to controls except for top bar and **Visualizer** panel controls.

At this point, it appears that the easiest way to get out of this condition is to disconnect from the machine and then reconnect to it though one can also stop and restart the application.  The subsequent homing operation is successful.

This is sub-optimal behavior (a bug).  Avoiding this condition is suggested.  The solution appears to be as simple as greying out the home button.  If one clicks the **Click to Unlock Machine** button before clicking the **Home** button, the homing operation proceeds as it should.

kbulgrien, 2024-01-20:
The problem condition does not appear to occur if the Z-axis is _already_ in the Z-limit when the application is first started up.  In other words, if the application is stopped and restarted with the axis in the Z-limit, upon restarting the application, it does not appear to matter if one clicks the **Home** button before or after activating the **Click to Unlock Machine** button.

SienciLabs, 2024-11-06:
Very interesting condition @kbulgrien, thanks for reporting

SienciLabs, 2024-11-06:
Is this behaviour still present in the latest 1.4.10 version of gSender?